osrf/ros2_test_cases #656: Executables in `gazebo_ros_pkgs`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `gazebo --verbose /opt/ros/rolling/share/gazebo_plugins/worlds/gazebo_ros_camera_distortion_barrel_demo.world`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run gazebo_ros_pkgs gazebo --verbose /opt/ros/rolling/share/gazebo_plugins/worlds/gazebo_ros_camera_distortion_barrel_demo.world
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a default Windows 11 binary installation with fastDDS

**_ERROR_**: the gazebo_ros_pkgs is not part of he binary.
But this is the same on a `jammy`binary installation so seems expected.

[image]

Yadunund, 2023-05-08:
Looks like `gazebo_ros_pkgs` is not available along with binaries. It's only available via debian or source. Might be good to have another workspace with this package built from source to continue with testing

Yadunund, 2023-05-11:
Turns out this test is meaningless on windows as gazebo is not a supported platform here. Sorry about that

closed 2023-05-11 by Yadunund