osrf/ros2_test_cases #657: Executables in `gazebo_ros_pkgs`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `gazebo --verbose /opt/ros/rolling/share/gazebo_plugins/worlds/gazebo_ros_camera_distortion_barrel_demo.world`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run gazebo_ros_pkgs gazebo --verbose /opt/ros/rolling/share/gazebo_plugins/worlds/gazebo_ros_camera_distortion_barrel_demo.world
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

iuhilnehc-ynos, 2023-05-08:
I changed the command to test.

- [ ] Check `gazebo --verbose /opt/ros/${ROS_DISTRO}/share/gazebo_plugins/worlds/gazebo_ros_camera_distortion_barrel_demo.world`

Use `ros2 launch gazebo_ros gazebo.launch.py verbose:=true world:=/opt/ros/${ROS_DISTRO}/share/gazebo_plugins/worlds/gazebo_ros_camera_distortion_barrel_demo.world` instead of `ros2 run gazebo_ros_pkgs gazebo --verbose /opt/ros/rolling/share/gazebo_plugins/worlds/gazebo_ros_camera_distortion_barrel_demo.world`

[image]
[image]

Yadunund, 2023-05-11:
Thanks for testing this!

closed 2023-05-11 by Yadunund