osrf/ros2_test_cases #660: Executables in `image_tools`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `cam2image`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run image_tools cam2image
       ```
  
    
  </details>
- [ ] **Check `showimage`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run image_tools showimage
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a default Windows 11 binary installation with FastDDS

Checked the 2 executables are correctly launching.
All ok, even tested sending image from my laptop for checking show_image, as I don't have cam on the Windows workstation.

[image]

clalancette, 2023-05-05:
Thank you for testing!

closed 2023-05-05 by clalancette