osrf/ros2_test_cases #661: Executables in `image_tools`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `cam2image`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run image_tools cam2image
       ```
  
    
  </details>
- [ ] **Check `showimage`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run image_tools showimage
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jjd9, 2023-05-06:
I would like to do this one please.

Terminal 1
```
ros2 run image_tools showimage --ros-args -p show_image:=true --ros-args -p window_name:="VeggieBurgers" --ros-args -p reliability:=best_effort
```

Terminal 2:
```
ros2 run image_tools cam2image --ros-args -p burger_mode:=true --ros-args -p reliability:=best_effort --ros-args -p width:=640 --ros-args -p height:=480
```

Image displayed in opencv GUI:
[image]

These executable's both appear to work properly.

One oddity (not really a bug, just something i found odd during testing with my webcam) is that that the cv::Mat adapter these executable's use doesnt preserve channel ordering. (e.g. if you have `cam2image` publish your webcam video and visualize it using `showimage`, `cam2image` is publishing a BGR image but `showimage` assumes any CV_8UC3 image it receives is RGB.

clalancette, 2023-05-06:
> One oddity (not really a bug, just something i found odd during testing with my webcam) is that that the cv::Mat adapter these executable's use doesnt preserve channel ordering. (e.g. if you have `cam2image` publish your webcam video and visualize it using `showimage`, `cam2image` is publishing a BGR image but `showimage` assumes any CV_8UC3 image it receives is RGB (since all the cv::Mat message conveys is that the image is CV_8UC3).

Hm, that does sound like a bit of a bug.  Could you please open up an issue at https://github.com/ros2/demos describing this issue?  We probably won't have time to fix it for Iron, but it is something we should track.

jjd9, 2023-05-06:
> Hm, that does sound like a bit of a bug. Could you please open up an issue at https://github.com/ros2/demos describing this issue? We probably won't have time to fix it for Iron, but it is something [...]

Sure! Issue posted: https://github.com/ros2/demos/issues/612

clalancette, 2023-05-07:
> Sure! Issue posted: [ros2/demos#612](https://github.com/ros2/demos/issues/612)

Thank you, it is much appreciated.

Given that, I'm going to close this out.  Thank you for testing!

closed 2023-05-07 by clalancette