osrf/ros2_test_cases #664: Executables in `intra_process_demo`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `camera_node`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run intra_process_demo camera_node
       ```
  
    
  </details>
- [ ] **Check `watermark_node`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run intra_process_demo watermark_node
       ```
  
    
  </details>
- [ ] **Check `image_view_node`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run intra_process_demo image_view_node
       ```
  
    
  </details>
- [ ] **Check `cyclic_pipeline`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run intra_process_demo cyclic_pipeline
       ```
  
    
  </details>
- [ ] **Check `image_pipeline_all_in_one`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run intra_process_demo image_pipeline_all_in_one
       ```
  
    
  </details>
- [ ] **Check `image_pipeline_with_two_image_view`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run intra_process_demo image_pipeline_with_two_image_view
       ```
  
    
  </details>
- [ ] **Check `two_node_pipeline`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run intra_process_demo two_node_pipeline
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a default Windows 11 binary installation with fastDDS.

Checked that all 7 executables are launching.
All ok

[image]

clalancette, 2023-05-05:
Thank you for testing!

closed 2023-05-05 by clalancette