osrf/ros2_test_cases #665: Executables in `intra_process_demo`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `camera_node`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run intra_process_demo camera_node
       ```
  
    
  </details>
- [ ] **Check `watermark_node`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run intra_process_demo watermark_node
       ```
  
    
  </details>
- [ ] **Check `image_view_node`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run intra_process_demo image_view_node
       ```
  
    
  </details>
- [ ] **Check `cyclic_pipeline`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run intra_process_demo cyclic_pipeline
       ```
  
    
  </details>
- [ ] **Check `image_pipeline_all_in_one`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run intra_process_demo image_pipeline_all_in_one
       ```
  
    
  </details>
- [ ] **Check `image_pipeline_with_two_image_view`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run intra_process_demo image_pipeline_with_two_image_view
       ```
  
    
  </details>
- [ ] **Check `two_node_pipeline`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run intra_process_demo two_node_pipeline
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-09:
On a debian jammy installation with fastDDS.

I have run the 7 commands of the testcase.

All OK.

[image]
[image]

[image]
[image]
[image]

clalancette, 2023-05-09:
Thank you for testing!

closed 2023-05-09 by clalancette