osrf/ros2_test_cases #667: Executables in `lifecycle`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `lifecycle_demo.launch.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch lifecycle lifecycle_demo.launch.py
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
The actual launch file found is ```ros2 launch lifecycle lifecycle_demo_launch.py```. That works as expected. Toggles between active and inactive. The check has passed.

Yadunund, 2023-05-09:
Thanks for testing!

closed 2023-05-09 by Yadunund