osrf/ros2_test_cases #668: Executables in `lifecycle`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `lifecycle_demo.launch.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch lifecycle lifecycle_demo.launch.py
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On default Windows 11 binary installation with FastDDS.

Executable is available and ok.

[image]

clalancette, 2023-05-05:
Thank you for testing!

closed 2023-05-05 by clalancette