osrf/ros2_test_cases #671: Executables in `pendulum_control`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `pendulum_demo`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_demo
       ```
  
    
  </details>
- [ ] **Check `pendulum_teleop`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_teleop
       ```
  
    
  </details>
- [ ] **Check `pendulum_launch.bash`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch pendulum_control pendulum_launch.bash
       ```
  
    
  </details>
- [ ] **Check `pendulum_logger`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_logger
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
There is not launch and executable files found  in pendulum_control package.

pradyum, 2023-05-10:
[image]

All the pendulum executables are in ```/opt/ros/iron/bin``` and not in pendulum_control package. The pendulum_logger, pendulum_demo, pendulum_teleop, pendulum_launch.bash are executing and all checks are passed.

Yadunund, 2023-05-10:
Thanks for testing. I've opened at ticket upstream. The targets are likely not installed correctly. I'll close this for now.

closed 2023-05-10 by Yadunund