osrf/ros2_test_cases #672: Executables in `pendulum_control`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `pendulum_demo`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_demo
       ```
  
    
  </details>
- [ ] **Check `pendulum_teleop`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_teleop
       ```
  
    
  </details>
- [ ] **Check `pendulum_launch.bash`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch pendulum_control pendulum_launch.bash
       ```
  
    
  </details>
- [ ] **Check `pendulum_logger`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_logger
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a default Windows 11 binary installation with fast DDS.

**_ERROR_**: The pendulum_demo is not present in the binary.

As per same test on `humble` (https://github.com/osrf/ros2_test_cases/issues/190)  , this package is only built for linux as per https://github.com/ros2/demos/blob/b64830e041dcd6bcd934cf222b53571f84706f6f/pendulum_control/CMakeLists.txt#L5-L8

Would be nice for future releases that this test-case can NOT be generated for windows. (if possible).

[image]

Yadunund, 2023-05-08:
Thanks for catching this! Opened this PR to not include this test for the next iteration https://github.com/audrow/yatm/pull/357

closed 2023-05-08 by Yadunund