osrf/ros2_test_cases #673: Executables in `pendulum_control`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `pendulum_demo`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_demo
       ```
  
    
  </details>
- [ ] **Check `pendulum_teleop`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_teleop
       ```
  
    
  </details>
- [ ] **Check `pendulum_launch.bash`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch pendulum_control pendulum_launch.bash
       ```
  
    
  </details>
- [ ] **Check `pendulum_logger`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run pendulum_control pendulum_logger
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-09:
On a debian jammy installation with fastDDS.

**_Error_**: The `pendulum_control` package does NOT provide executables through `ros2 run`.
The executable are provided directly on the command-line.

But the 4 commands work when launched directly  from command line:

[image]
[image]

Yadunund, 2023-05-09:
Nice find! I've opened this ticket upstream https://github.com/ros2/demos/issues/622

I'll close this ticket.

closed 2023-05-09 by Yadunund