osrf/ros2_test_cases #674: Executables in `quality_of_service_demo_cpp`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `lifespan`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp lifespan
       ```
  
    
  </details>
- [ ] **Check `liveliness`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp liveliness
       ```
  
    
  </details>
- [ ] **Check `deadline`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp deadline
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-08:
Hello, could you please assign this task to me?

juliamp22, 2023-05-08:
Check **lifespan** works fine:

[image]

juliamp22, 2023-05-08:
Check **liveliness** works fine:

[image]

juliamp22, 2023-05-08:
Check **deadline** works fine:

[image]

juliamp22, 2023-05-08:
I think that this issue can be closed, as all works! Thank you so much.

Yadunund, 2023-05-09:
Great, thanks for testing!

closed 2023-05-09 by Yadunund