osrf/ros2_test_cases #677: Executables in `quality_of_service_demo_cpp`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `lifespan`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp lifespan
       ```
  
    
  </details>
- [ ] **Check `liveliness`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp liveliness
       ```
  
    
  </details>
- [ ] **Check `deadline`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp deadline
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

hwoithe, 2023-05-07:
- [x] Check `lifespan`
```
$ ros2 run quality_of_service_demo_cpp lifespan
Usage for lifespan:
lifespan lifespan_duration [--history history_depth] [--publish-count publish_count] [--subscribe-after subscribe_after] [-h]
required arguments:
lifespan duration: Duration in positive integer milliseconds of the Lifespan QoS setting.
optional arguments:
-h : Print this help message.
--history history : The depth of the Publisher's history queue - the maximum number of messages it will store for late-joining subscriptions. Defaults to 10
--publish-count publish_n_messages : How many messages to publish before stopping. Defaults to 5
--subscribe-after subscribe_after_duration : The Subscriber will be created this long in positive integer milliseconds after application startup. Defaults to 5000
```

- [x] Check `liveliness`
```
$ ros2 run quality_of_service_demo_cpp liveliness
Usage for liveliness:
liveliness lease_duration [--policy liveliness_policy] [--topic-assert-period topic_assert_liveliness_period] [-h]
required arguments:
lease_duration: Duration in positive integer milliseconds after which an inactive Publisher is considered not-alive. 0 means never.
optional arguments:
-h : Print this help message.
--policy liveliness_policy : You may specify AUTOMATIC, or MANUAL_BY_TOPIC. Defaults to AUTOMATIC
--topic-assert-period topic_assert_period : How often the Publisher will assert the liveliness of its Publisher , in positive integer milliseconds. Defaults to 0 (never)
--kill-publisher-after kill_publisher_after : Kill the publisher after this amount of time (in uint milliseconds). In AUTOMATIC - destroy the whole node. In MANUAL_BY_TOPIC, stop topic liveliness assertion. Defaults to 3000
```

- [x] Check deadline
```
$ ros2 run quality_of_service_demo_cpp deadline
Usage for deadline:
deadline deadline_duration [--publish-for publish_for] [--pause-for pause_for] [-h]
required arguments:
deadline_duration: Duration in positive integer milliseconds of the Deadline QoS setting.
optional arguments:
-h : Print this help message.
--publish-for publish_for : Duration to publish (in positive integer milliseconds) until pausing the talker. Defaults to 5000.
--pause-for pause_for : Duration to pause (in positive integer milliseconds) before starting to publish again. Defaults to 1000.
```

All of these are runnable.

---

`lifespan` test with argument

```
# terminal 1
$ ros2 run quality_of_service_demo_cpp lifespan 12345
[INFO] [1683462810.200801484] [talker]: Talker starting up
[INFO] [1683462810.201412608] [listener]: Listener starting up
[INFO] [1683462810.701636688] [talker]: Publishing: 'Talker says 0'
[INFO] [1683462811.201634949] [talker]: Publishing: 'Talker says 1'
[INFO] [1683462811.701643490] [talker]: Publishing: 'Talker says 2'
[INFO] [1683462812.201596771] [talker]: Publishing: 'Talker says 3'
[INFO] [1683462812.701642323] [talker]: Publishing: 'Talker says 4'
[INFO] [1683462815.202967549] [listener]: Listener heard: [Talker says 0]
[INFO] [1683462815.203093222] [listener]: Listener heard: [Talker says 1]
[INFO] [1683462815.203169114] [listener]: Listener heard: [Talker says 2]
[INFO] [1683462815.203241869] [listener]: Listener heard: [Talker says 3]
[INFO] [1683462815.203315133] [listener]: Listener heard: [Talker says 4]

# terminal 2
$ ros2 topic info /qos_lifespan_chatter -v
Type: std_msgs/msg/String

Publisher count: 1

Node name: talker
[... 142 lines cut ...]
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: 12345000000 nanoseconds
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: listener
Node namespace: /
Topic type: std_msgs/msg/String
Topic type hash: UNKNOWN
Endpoint type: SUBSCRIPTION
GID: 01.0f.96.84.e9.6b.68.62.00.00.00.00.00.00.22.04
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: 12345000000 nanoseconds
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```

clalancette, 2023-05-07:
Thank you for testing!

closed 2023-05-07 by clalancette