osrf/ros2_test_cases #680: Executables in `quality_of_service_demo_py`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `lifespan`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py lifespan
       ```
  
    
  </details>
- [ ] **Check `liveliness`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py liveliness
       ```
  
    
  </details>
- [ ] **Check `deadline`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py deadline
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a default Windows 11 binary installation with FastDDS.

Checks all 3 executables availability of the quality_of_service_demo_py package.
All Ok

[image]

Yadunund, 2023-05-05:
Thanks for testing!

closed 2023-05-05 by Yadunund