osrf/ros2_test_cases #682: Executables in `rosbag2`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `ros2 bag record -a`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rosbag2 ros2 bag record -a
       ```
  
    
  </details>
- [ ] **Check `ros2 bag record <topic1> <topic2> <topic3>`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rosbag2 ros2 bag record <topic1> <topic2> <topic3>
       ```
  
    
  </details>
- [ ] **Check `ros2 bag info <bag_file>`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rosbag2 ros2 bag info <bag_file>
       ```
  
    
  </details>
- [ ] **Check `ros2 bag play <bag_file>`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rosbag2 ros2 bag play <bag_file>
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-08:
# Setup

DDS vendor: Cyclone
BuildType: Binary
Os: Windows
Chip: Amd64

[image]

# Result

## Check 1
[image]

## Check 2

[image]

## Check 3
[image]

## Check 4

[image]

**All Checks are passed.**

clalancette, 2023-05-08:
Thank you for testing!

closed 2023-05-08 by clalancette