osrf/ros2_test_cases #684: Executables in `rosbag2`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `ros2 bag record -a`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rosbag2 ros2 bag record -a
       ```
  
    
  </details>
- [ ] **Check `ros2 bag record <topic1> <topic2> <topic3>`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rosbag2 ros2 bag record <topic1> <topic2> <topic3>
       ```
  
    
  </details>
- [ ] **Check `ros2 bag info <bag_file>`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rosbag2 ros2 bag info <bag_file>
       ```
  
    
  </details>
- [ ] **Check `ros2 bag play <bag_file>`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rosbag2 ros2 bag play <bag_file>
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a default Window binary installation with FastDDS.

I checked that the 4 commands are working.

**_WARNING_**: They are not working as displayed in each **Try** sections, the commands in this test case should start with `ros2 bag` not `ros2 run rosbag2 ros2 bag`

[image]

Yadunund, 2023-05-05:
Thanks for testing this!

I've opened this PR https://github.com/audrow/yatm/pull/352 to fix the input commands for the test.

closed 2023-05-05 by Yadunund