osrf/ros2_test_cases #685: Executables in `rosbag2`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `ros2 bag record -a`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rosbag2 ros2 bag record -a
       ```
  
    
  </details>
- [ ] **Check `ros2 bag record <topic1> <topic2> <topic3>`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rosbag2 ros2 bag record <topic1> <topic2> <topic3>
       ```
  
    
  </details>
- [ ] **Check `ros2 bag info <bag_file>`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rosbag2 ros2 bag info <bag_file>
       ```
  
    
  </details>
- [ ] **Check `ros2 bag play <bag_file>`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rosbag2 ros2 bag play <bag_file>
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
The bag CLI is working as expected. I am avoid the text generation error which produced this test commands.

Yadunund, 2023-05-11:
Great, thanks for testing!

closed 2023-05-11 by Yadunund