osrf/ros2_test_cases #686: Executables in `rqt`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `rqt`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rqt rqt
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-08:
# Setup

- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

[image]

# Result

[image]

**No executable found, the issue is added to discussion here https://github.com/osrf/ros2_test_cases/discussions/1048**

Yadunund, 2023-05-09:
Thanks for reporting. Does running `rqt` directly work?

pradyum, 2023-05-09:
yeah the rqt runs directly

closed 2023-05-09 by Yadunund