osrf/ros2_test_cases #687: Executables in `rqt`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `rqt`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rqt rqt
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
The rqt package is not while installing with ```ros-iron-desktop-full```. If I install rqt separately though ```sudo apt-get install ros-iron-rqt```, the executable "rqt" is not present in the package.

pradyum, 2023-05-11:
There is no rqt executable in the rqt package. It works as a standalone command ```rqt```.

Yadunund, 2023-05-16:
Thanks for trying this. It's a known issue and being worked on here https://github.com/ros-visualization/rqt/issues/270
So i'll close this ticket.

closed 2023-05-16 by Yadunund