osrf/ros2_test_cases #688: Executables in `rqt`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `rqt`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rqt rqt
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a default Windows 11 binary installation with FastDDS

**_ERROR_**: This is not working as expected, rqt binary is not available through the `ros2 run rqt rqt` command.

`rqt` executable itself is available.

This seems to have been already reported for `humble` https://github.com/ros-visualization/rqt/issues/270, and remain unchanged in `iron`.

[image]

Yadunund, 2023-05-08:
Thanks for highlighting, i've added it to the list of rqt issues here https://github.com/osrf/ros2_test_cases/discussions/1048

Do you mind testing by running just `rqt` instead? I've opened this PR https://github.com/audrow/yatm/pull/356 to update the test case but hopefully by the next time the bug has been fixed upstream.

BorisBoutillier, 2023-05-08:
just running `rqt` is fine, and works properly.

Yadunund, 2023-05-09:
Thanks for confirming!

closed 2023-05-09 by Yadunund