osrf/ros2_test_cases #689: Executables in `rqt`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `rqt`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rqt rqt
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Kartik9250, 2023-05-06:
Hey, I would like to take on this ticket

Kartik9250, 2023-05-06:
- Running the given command gives out an error saying that `No executable found`
[image]

- I tried listing the ros2 executables:
[image]

- Also the auto-completion does list it out as an option, but it not running for some reason:
[image]

Yadunund, 2023-05-08:
This is a known bug with rqt https://github.com/ros-visualization/rqt/issues/270

We will aim to fix this in the future. In the meanwhile, could you run the command `rqt` instead?

Yadunund, 2023-05-18:
We've updated the test case to fix the input in the future https://github.com/audrow/yatm/pull/374

I've tested with just `rqt` and it works.

closed 2023-05-18 by Yadunund