osrf/ros2_test_cases #690: Executables in `rttest`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `rttest_plot`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rttest rttest_plot
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-08:
Hello, could you please assign this Issue to me?

juliamp22, 2023-05-08:
Hello, It cannot also find the package: 

[image]

Should I install or download a specific package? Thank you so much.