osrf/ros2_test_cases #692: Executables in `rttest`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `rttest_plot`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rttest rttest_plot
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a default Windows 11 binary installation with Fast DDS

**_ERROR_**: This testcase is not working as expected, the `rttest` pkg is not provided in the windows binary installation.
I have checked on a jammy binary instalation, it is present there.

I am not sure if this absence is expected from the windows binary. If this is expected, this test_case should be removed.

[image]

Yadunund, 2023-05-05:
This PR will ensure such tests are not created in the future https://github.com/audrow/yatm/pull/351

closed 2023-05-05 by Yadunund