osrf/ros2_test_cases #693: Executables in `rttest`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `rttest_plot`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rttest rttest_plot
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-09:
On a debian jammy installation with fastDDS.

I have been able to check the rttest_plot command with help from the "[Understanding Real-time programming](https://docs.ros.org/en/iron/Tutorials/Demos/Real-Time-Programming.html#plotting-results)" tutorial page.

I would suggest to at least reference this page in this testcase, as the provided command line in this testcase will just fail with 
```
:ros-iron-ws$ ros2 run rttest rttest_plot
usage: rttest_plot [-h] [-s] [-o OUTFILE] filename
rttest_plot: error: the following arguments are required: filename
[ros2run]: Process exited with failure 2
```

[image]

Yadunund, 2023-05-11:
Thanks for trying this out. I've noted the need to include a link to the doc page in the test case.

closed 2023-05-11 by Yadunund