osrf/ros2_test_cases #694: Executables in `rviz2`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `rviz2`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rviz2 rviz2
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-08:
# Setup

- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

[image]

**Additional Changes: The zlib1.dll is placed at rviz_ogre_vendor/bin folder.**
This issue is rectified here https://github.com/ros2/ros2_documentation/pull/3542

# Result

[image]

The rviz2 is executing as expected. The check is passed.

Yadunund, 2023-05-09:
Thanks for testing!

closed 2023-05-09 by Yadunund