osrf/ros2_test_cases #697: Executables in `rviz2`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `rviz2`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run rviz2 rviz2
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Kartik9250, 2023-05-06:
[image]
[image]

- The Error that you are seeing shows up after adding the `/scan/laserscan` onto `rviz2`
- As per the required test case it launches just fine.

Yadunund, 2023-05-08:
Thanks for testing! I'm assuming that error printed once at the time of subscription since the `LaserScan` display in Rviz is not in error state. This could have been due to some initial subscription overhead which caused a delay in receiving the message.

closed 2023-05-08 by Yadunund