osrf/ros2_test_cases #698: Executables in `tf2_ros`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `tf2_echo`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros tf2_echo
       ```
  
    
  </details>
- [ ] **Check `tf2_monitor`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros tf2_monitor
       ```
  
    
  </details>
- [ ] **Check `static_transform_publisher`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros static_transform_publisher
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-08:
# Setup

- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

[image]

# Result

[image]

**The executable in ```tf2_ros``` is working. The check is passed.**

clalancette, 2023-05-08:
Thank you for testing!

closed 2023-05-08 by clalancette