osrf/ros2_test_cases #699: Executables in `tf2_ros`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `tf2_echo`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros tf2_echo
       ```
  
    
  </details>
- [ ] **Check `tf2_monitor`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros tf2_monitor
       ```
  
    
  </details>
- [ ] **Check `static_transform_publisher`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros static_transform_publisher
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-09:
**The tf2_ros node are working as expected with their arguments. The checks have passed.**

Yadunund, 2023-05-09:
Thanks for testing!

closed 2023-05-09 by Yadunund