osrf/ros2_test_cases #701: Executables in `tf2_ros`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `tf2_echo`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros tf2_echo
       ```
  
    
  </details>
- [ ] **Check `tf2_monitor`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros tf2_monitor
       ```
  
    
  </details>
- [ ] **Check `static_transform_publisher`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tf2_ros static_transform_publisher
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Kartik9250, 2023-05-06:
Hey, I would like to work on this issue

Kartik9250, 2023-05-06:
### Checks

- [x] Check `tf2_echo`
[image]

- [x] Check `tf2_monitor`
[image]

- [x] Check `tf2_transform_publisher`
[image]

<br>

I am probably not running the test cases properly. Could you help me out

Yadunund, 2023-05-08:
Thanks for testing, I've opened this PR to update the test case in the future https://github.com/audrow/yatm/pull/358
Could you run the test with `ros2 run tf2_ros static_transform_publisher --frame-id world --child-frame-id map`?

Yadunund, 2023-05-18:
Closing this after testing.

closed 2023-05-18 by Yadunund