osrf/ros2_test_cases #702: Executables in `tlsf_cpp`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `tlsf_allocator_example`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tlsf_cpp tlsf_allocator_example
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-08:
Hello, could you please assign this Issue to me?

juliamp22, 2023-05-08:
The package cannot be found:

[image]

Should I do something? Install a package or a dependency? Thank you so much.

Yadunund, 2023-05-09:
Sorry for the trouble but this test is not valid as Windows does not support realtime functionality. We updated our test tool to prevent such tickets from being generated in the future https://github.com/audrow/yatm/pull/351

closed 2023-05-09 by Yadunund