osrf/ros2_test_cases #704: Executables in `tlsf_cpp`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `tlsf_allocator_example`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tlsf_cpp tlsf_allocator_example
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a default Windows 11 binary installation with FastDDS.

**_ERROR_**: This is not working as expected by this test, the `tlsf` and `tlsf_cpp` packages are not provided in the windows binary installation.
I have checked on a jammy binary installation, and these two packages are available.

Not sure if this is expected or not in the windows binary. If this is expected, then this test_case should be removed.

Screenshot showing it is not in my `ros2 pkg list` :
[image]

Yadunund, 2023-05-05:
Yeah I don't think we have realtime support on Windows.
@clalancette does that make sense? If so I'll update `yatm`

clalancette, 2023-05-05:
> Yeah I don't think we have realtime support on Windows.
> @clalancette / @mjcarroll does that make sense? If so I'll update `yatm`

Correct, we don't have realtime support on Windows.  We should remove this test case for Windows in YATM.

Yadunund, 2023-05-05:
I've opened this PR which should avoid such tests being created in the future.

closed 2023-05-05 by Yadunund