osrf/ros2_test_cases #705: Executables in `tlsf_cpp`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Check `tlsf_allocator_example`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tlsf_cpp tlsf_allocator_example
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Kartik9250, 2023-05-06:
I would like to work on this ticket

Kartik9250, 2023-05-06:
- I tried the given command `ros2 run tlsf_cpp tlsf_allocator_example`.
  The terminal prints out saying that `Intra-process pipeline is OFF.`

- Using `ros2 run tlsf_cpp tlsf_allocator_example intra`,  gives out: `Intra-process pipeline is ON`

- Not sure if this is all that was needed to be tested. Found some [old issue](https://github.com/ros2/demos/issues/292)
so I tried that as well:
[image]
    - I also echo-ed the topic and the `data` iterated till `1912`.

<br>

- In case of `Intra-process pipeline OFF`:
```bash
Global new was called 17635 times during spin
Global delete was called 17336 times during spin
Allocator new was called 17238 times during spin
Allocator delete was called 17217 times during spin
```
- Whereas in case of `Intra-process pipeline ON`:
```bash
Global new was called 14818 times during spin
Global delete was called 14503 times during spin
Allocator new was called 17235 times during spin
Allocator delete was called 17216 times during spin
```

clalancette, 2023-05-06:
Yep, this all looks correct.  Thank you for testing!

closed 2023-05-06 by clalancette