osrf/ros2_test_cases #710: Installation instructions

Yadunund, 2023-04-05:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- DDS vendor: Connext
- BuildType: Binary
- Chip: Amd64
- Os: Ubuntu Jammy

## Links
- [Installation instructions](https://docs.ros.org/en/rolling/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-07:
**Setup**

- DDS vendor: Connext
- BuildType: Binary
- Chip: Amd64
- Os: Ubuntu Jammy

[image]

The installation works. But the distribution shows as rolling.

clalancette, 2023-05-08:
> The installation works. But the distribution shows as rolling.

Yeah, we'll fix that on the next round of building the binaries.

Thanks for testing!