osrf/ros2_test_cases #711: Installation instructions

Yadunund, 2023-04-05:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- DDS vendor: Connext
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Installation instructions](https://docs.ros.org/en/rolling/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

mjcarroll, 2023-05-03:
Confirmed working for me. MSVC 2022, Windows 11, followed installation instructions.

I'm going to leave this unassigned because I have been doing enough Windows work lately that a second set of eyes on these instructions wouldn't hurt.

luca-della-vedova, 2023-05-09:
Tested on Windows 10, using an evaluation license for Connext:

Minor docs issue, version 6.1.1 has been released and can be downloaded from the RTI website, while [docs ](https://docs.ros.org/en/iron/Installation/DDS-Implementations/Install-Connext-University-Eval.html) refer to 6.0.1

[image]

Installation works OK but running examples fails with the following error:

[image]

After clicking Ok the terminal prints an error loading the library:

[image]

I'm not sure whether this is an issue in the `rmw_connextdds` package or if it's due to a breaking change in 6.1.1, however I couldn't find any (obvious) way to download a 6.0.1 trial from the RTI website.

Since the issue is at the RMW level and a basic demo doesn't work I doubt any other application will work

mjcarroll, 2023-05-09:
I ran into the same issue, presumably because of a version mismatch. As a datapoint, when building from source against 6.1.1, everything seems to work.

Yadunund, 2023-05-10:
To follow up on yesterday's discussion, the issue might be ABI incompatibility between the 6.0.1 and 6.1.1. The Windows binaries are built against 6.0.1.

Yadunund, 2023-05-18:
We will be adding this issue to the list of "Known Issues" at release. Building from source is the alternative.

closed 2023-05-18 by Yadunund