osrf/ros2_test_cases #713: Installation instructions

Yadunund, 2023-04-05:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- DDS vendor: Connext
- BuildType: Debian
- Chip: Amd64
- Os: Ubuntu Jammy

## Links
- [Installation instructions](https://docs.ros.org/en/rolling/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jjd9, 2023-05-04:
May I take this one?

jjd9, 2023-05-04:
Yay! Worked seamlessly :smile: 
Went through the Debian instructions on my machine: https://docs.ros.org/en/iron/Installation/Ubuntu-Install-Debians.html
Setup the DDS: https://docs.ros.org/en/iron/How-To-Guides/Working-with-multiple-RMW-implementations.html
(although the instructions arent explicit about the fact that you need to apt install `ros-iron-rmw-connextdds`).
Then I ran the connextdds examples. here's the output
Terminal 1:
```
johnd@johnd-X510UQ:~$ source /opt/ros/iron/setup.bash
johnd@johnd-X510UQ:~$ RMW_IMPLEMENTATION=rmw_connextdds ros2 run demo_nodes_cpp talker
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00 See www.rti.com for more information.
[INFO] [1683159650.731320356] [talker]: Publishing: 'Hello World: 1'
[INFO] [1683159651.731285056] [talker]: Publishing: 'Hello World: 2'
[INFO] [1683159652.731115675] [talker]: Publishing: 'Hello World: 3'
...
```

Terminal 2:
```
johnd@johnd-X510UQ:~$ source /opt/ros/iron/setup.bash
johnd@johnd-X510UQ:~$ RMW_IMPLEMENTATION=rmw_connextdds ros2 run demo_nodes_py listener
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00 See www.rti.com for more information.
[INFO] [1683159669.780425101] [listener]: I heard: [Hello World: 20]
[INFO] [1683159670.734953813] [listener]: I heard: [Hello World: 21]
[INFO] [1683159671.733945256] [listener]: I heard: [Hello World: 22]
[INFO] [1683159672.733670005] [listener]: I heard: [Hello World: 23]
[INFO] [1683159673.734492868] [listener]: I heard: [Hello World: 24]
[INFO] [1683159674.734437475] [listener]: I heard: [Hello World: 25]
[INFO] [1683159675.734368252] [listener]: I heard: [Hello World: 26]
```

jjd9, 2023-05-04:
here is evidence that my system is amd64 + jammy
```
johnd@johnd-X510UQ:~$ lsb_release -a
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.2 LTS
Release:	22.04
Codename:	jammy
johnd@johnd-X510UQ:~$ lscpu
Architecture:            x86_64
```

fujitatomoya, 2023-05-04:
working on arm64 and amd64 together.

fujitatomoya, 2023-05-04:
Error observed during demo talker and listener.

```bash
root@tomoyafujita:~# RMW_IMPLEMENTATION=rmw_connextdds ros2 run demo_nodes_cpp talker
[ERROR] [1683177387.129087956] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_connextdds'), with error message 'failed to load shared library 'librmw_connextdds.so' due to dlopen error: librmw_connextdds.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99, at ./src/functions.cpp:65', exiting with 1., at ./src/rcl/rmw_implementation_identifier_check.c:139

[ros2run]: Process exited with failure 1
```
the reason is there is no `ros-iron-rmw-connextdds` debian package is installed with procedure.

```bash
root@tomoyafujita:~# apt install ros-iron-rmw-connextdds
```

then,

```bash
root@tomoyafujita:~# RMW_IMPLEMENTATION=rmw_connextdds ros2 run demo_nodes_cpp talker
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00 See www.rti.com for more information.
COMMENDSrWriterService_assertRemoteReader:Discovered remote reader with GUID 0X10F60DB,0XDA25FD13,0,0X200C7 using a non-addressable locator.
This can occur if the transport is not installed and/or enabled in the local participant.
See https://community.rti.com/kb/what-does-cant-reach-locator-error-message-mean for additional info.
can't reach: locator: 16://5560:DB00:0000:0000:0000:0000:0000:0000:7414
COMMENDSrWriterService_assertRemoteReader:Discovered remote reader with GUID 0X10F60DB,0XDA25FD13,0,0X3C7 using a non-addressable locator.
This can occur if the transport is not installed and/or enabled in the local participant.
See https://community.rti.com/kb/what-does-cant-reach-locator-error-message-mean for additional info.
can't reach: locator: 16://5560:DB00:0000:0000:0000:0000:0000:0000:7414
COMMENDSrWriterService_assertRemoteReader:Discovered remote reader with GUID 0X10F60DB,0XDA25FD13,0,0X4C7 using a non-addressable locator.
This can occur if the transport is not installed and/or enabled in the local participant.
See https://community.rti.com/kb/what-does-cant-reach-locator-error-message-mean for additional info.
can't reach: locator: 16://5560:DB00:0000:0000:0000:0000:0000:0000:7414
COMMENDSrWriterService_assertRemoteReader:Discovered remote reader with GUID 0X10F60DB,0XDA25FD13,0,0X204 using a non-addressable locator.
This can occur if the transport is not installed and/or enabled in the local participant.
See https://community.rti.com/kb/what-does-cant-reach-locator-error-message-mean for additional info.
can't reach: locator: 16://5560:DB00:0000:0000:0000:0000:0000:0000:7415
[INFO] [1683177443.602486553] [talker]: Publishing: 'Hello World: 1'
[INFO] [1683177444.602454236] [talker]: Publishing: 'Hello World: 2'
[INFO] [1683177445.602417128] [talker]: Publishing: 'Hello World: 3'
[INFO] [1683177446.602458255] [talker]: Publishing: 'Hello World: 4'
[INFO] [1683177447.602428608] [talker]: Publishing: 'Hello World: 5'
^C[INFO] [1683177448.260345849] [rclcpp]: signal_handler(signum=2)
[D0000|DELETE Participant] PRESPsService_removeMatchingRemoteEndpoints:!precondition: "me == ((void *)0) || worker == ((void *)0) || remoteGuid == ((void *)0) || !( ((remoteGuid->prefix.hostId == (0x00000000)) && (remoteGuid->prefix.appId == (0x00000000)) && (remoteGuid->prefix.instanceId == (0x00000000)) && (remoteGuid->objectId != (0x00000000))) || ((remoteGuid->prefix.hostId != (0x00000000)) && (remoteGuid->prefix.appId != (0x00000000)) && (remoteGuid->prefix.instanceId != (0x00000000)) && (remoteGuid->objectId == (0x00000000))) )"
        Backtrace:
        #1      ?? ??:0 [0xA7B8D2E7]
        #2      ?? ??:0 [0xA799DF72]
        #3      ?? ??:0 [0xA797C127]
        #4      ?? ??:0 [0xA791A650]
        #5      ?? ??:0 [0xA791B97F]
        #6      ?? ??:0 [0xA7931C19]
        #7      ?? ??:0 [0xA7931801]
        #8      ?? ??:0 [0xA790BE5F]
        #9      ?? ??:0 [0xA7973BDE]
        #10     ?? ??:0 [0xA8670541]
        #11     ?? ??:0 [0xA863E1FC]
        #12     ?? ??:0 [0xA8F001F5]
        #13     ?? ??:0 [0xA8F3A74C]
        #14     ?? ??:0 [0xAA39C5F8]
        #15     ?? ??:0 [0xAA9AE309]
        #16     ?? ??:0 [0xAA9AE4A5]
        #17     ?? ??:0 [0xAA9746BA]
        #18     ?? ??:0 [0xAA9ADCF7]
        #19     ?? ??:0 [0xAA9ADDAA]
        #20     ?? ??:0 [0xAA9746BA]
        #21     ?? ??:0 [0xAA9AD7AF]
        #22     ?? ??:0 [0x523AA825]
        #23     ?? ??:0 [0xAA3EBD90]
        #24     ?? ??:0 [0xAA3EBE40]
        #25     ?? ??:0 [0x523AB0D5]
[D0000|DELETE Participant] DISCParticipantDiscoveryPlugin_removeRemoteParticipantI:!remove remote endpoints of remote participant: 0X10F60DB,0XDA25FD13,0,0X1C1
```

I am not even sure the above standart output is expected result.

john-maidbot, 2023-05-04:
> @jjd9 could I check if you installed the `rmw_connextdds` binary via `sudo apt install ros-iron-rmw-connextdds` prior to your testing?

@Yadunund yes, that is what I did prior to testing.

jjd9, 2023-05-04:
@fujitatomoya I think the problem is that you have a node already running that isnt using RTI connextdds. when I run two nodes with RTI connext dds, it runs fine. but if i run a talker on RTI connext dds and a listener on fast-dds, it gives the same warnings you mentioned and throws that error when i exit. i'll post an example. can you please try the demo again after verifying there are no lingering ros2 nodes running?

jjd9, 2023-05-04:
start listener on fastdds

start talker on RTI connext dds

fujitatomoya, 2023-05-04:
@jjd9 thanks for checking on that. yeah i was playing with multiple RMWs with containers, that might have led to that issue.

Kartik9250, 2023-05-05:
> Error observed during demo talker and listener.
> 
> ```shell
> root@tomoyafujita:~# RMW_IMPLEMENTATION=rmw_connextdds ros2 run demo_nodes_cpp talker
> [ERROR] [1683177387.129087956] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_connextdds'), with error message 'failed to load shared library 'librmw_connextdds.so' due to dlopen error: librmw_connextdds.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99, at ./src/functions.cpp:65', exiting with 1., at ./src/rcl/rmw_implementation_identifier_check.c:139

I went through the installation it worked just fine. I also did came across a similar error I think that was because earlier i was using ros2 humble & connextdds earlier and humble wasnt properly removed. I did a `sudo apt autoremove` and opened a new terminal. It worked after that.

fujitatomoya, 2023-05-09:
@Yadunund i created the follow-up issue https://github.com/ros2/ros2_documentation/issues/3573, with that we can follow-up and enhance the docs later. i think this test issue for iron can be closed.

Yadunund, 2023-05-10:
Awesome, thanks.

closed 2023-05-10 by Yadunund