osrf/ros2_test_cases #717: Installation instructions

Yadunund, 2023-04-05:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Installation instructions](https://docs.ros.org/en/rolling/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

bai-ivan, 2023-05-06:
Can't assign to me, but will test this now ;)

bai-ivan, 2023-05-06:
Using the provided release binary file 

installed choco
installed python
installed vcredist (i had 140 but not using "clean" windows so i probably had it from other stuff)
installed VS2019
installed opencv
installed cmake
Installed tinyxml2 v6.0.0
Installed asio v1.12.1
Installed cunit v2.1.3
Installed bullet v3.17
Installed tinyxml-usestl v2.6.2
Installed eigen v3.3.4

Suggestion: note about environment refresh / terminal reopen before installing pip/setuptools

installed pip deps

Note: Probably give out a note that installing QT requires an qt account. Made one, no issues with it.
installed QT

when calling ros2 over local_setup.bat i got this:

```
C:\Users\user\ros2_ws>call c:\dev\ros2-iron\local_setup.bat
"[rti_connext_dds_cmake_module][warning] RTI Connext DDS environment script not found (\resource\scripts\rtisetenv_x64Win64VS2017.bat). RTI Connext DDS will not be available at runtime, unless you already configured PATH manually."

```

Found no binary install of CycloneDDS for windows, so i tried via source, failed : 

C:\Users\user\ros2_ws>rosdep install --from src -i
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
cyclonedds: Unsupported OS [windows]
rmw_cyclonedds_cpp: Unsupported OS [windows]

If something else is needed on this, let me know.

pradyum, 2023-05-07:
# Setup

- DDS vendor: Cyclone
- BuildType: Binary
- Chip: Amd64
- Os: Windows

# Result

[image]

**The installation instruction works**

pradyum, 2023-05-07:
@bai-ivan the following output is just a warning and not an error. The RTI connext has to be installed separately. Since it is not there in PATH, this warning says that RTI connext cannot be used at runtime. 

While cyclonedds is available in it binary file.

[image]

bai-ivan, 2023-05-08:
I'm unable to find a reasonable installation instruction for cyclonedds for Windows (without wsl).
Probably if i invest half a day digging in google i can get it working, but i believe this is going out of scope, i assume we are testing here if a normal inexperienced user tries to follow the instructions.

Yadunund, 2023-05-08:
Thanks for testing! I was also able to verify cyclonedds installed via binaries on windows.

closed 2023-05-08 by Yadunund