osrf/ros2_test_cases #719: Installation instructions

Yadunund, 2023-04-05:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Chip: Amd64
- Os: Ubuntu Jammy

## Links
- [Installation instructions](https://docs.ros.org/en/rolling/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Abishalini, 2023-05-03:
Followed the instructions to install Iron debian, set my DDS setting to cyclone and was able to successfully build my workspace.

clalancette, 2023-05-03:
Thank you for testing!

closed 2023-05-03 by clalancette