osrf/ros2_test_cases #720: Installation instructions

Yadunund, 2023-04-05:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Chip: Arm64
- Os: Ubuntu Jammy

## Links
- [Installation instructions](https://docs.ros.org/en/rolling/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

fujitatomoya, 2023-05-04:
/assign @fujitatomoya

fujitatomoya, 2023-05-04:
working on arm64 and amd64 together.

fujitatomoya, 2023-05-04:
### main installation procedure

- https://docs.ros.org/en/iron/Installation/Ubuntu-Install-Debians.html
- https://docs.ros.org/en/iron/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html

### platform

- aarch64 (arm64)
- amd64

### result

confirmed that https://docs.ros.org/en/iron/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html#run-the-talker-and-listener works w/o problems

Yadunund, 2023-05-04:
Thanks for testing!

closed 2023-05-04 by Yadunund