osrf/ros2_test_cases #722: Installation instructions

Yadunund, 2023-04-05:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Chip: Amd64
- Os: Ubuntu Jammy

## Links
- [Installation instructions](https://docs.ros.org/en/rolling/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

cdhabecker, 2023-05-03:
I'll do this one.

cdhabecker, 2023-05-03:
Submitted https://github.com/osrf/ros2_test_cases/issues/1046 "Install options in doc are inconsistently named".

cdhabecker, 2023-05-03:
Problems in https://docs.ros.org/en/iron/Installation/Alternatives/Ubuntu-Install-Binary.html
1. In Note, [ROS base variant](https://ros.org/reps/rep-2001.html#ros-base) links to Ardent Apalone section.
2. In Note, [ROS desktop variant](https://ros.org/reps/rep-2001.html#desktop-variants) links to a nonexistent anchor.
3. In "Install ROS 2", [releases page](https://github.com/ros2/ros2/releases) does not link to Iron section.
4. In "Install ROS 2", "Download the latest package for Ubuntu" -> "In the Assets section, download the lastest archive for Ubuntu".
5. In "Install ROS 2", "Note: there may be more than one binary download option which might cause the file name to differ." -> delete, unnecessary.
6. In "Install ROS 2", "Unpack it:" -> "Unpack the archive into a new directory. You can name the directory whatever you like. These instructions assume ~/ros2_iron."

Problems with the tester:
1. I failed to copy the shell output up through the "apt upgrade" step.

cdhabecker, 2023-05-04:
**SUCCESS**

I failed to copy the shell output up through the "apt upgrade" step. All the commands were nominal. This shell output resumes after the upgrade.
```
habecker@2023-05-03 16:38:54:~$ sudo apt upgrade
[sudo] password for habecker: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
habecker@2023-05-03 16:39:08:~$ lsb_release --all
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.2 LTS
Release:	22.04
Codename:	jammy
habecker@2023-05-03 16:52:20:~$ lscpu | grep Model
Model name:                      13th Gen Intel(R) Core(TM) i5-13500
Model:                           191
habecker@2023-05-03 16:52:32:~$ cd ros2_iron/
habecker@2023-05-03 16:52:53:~/ros2_iron$ ls
ros2-linux
habecker@2023-05-03 16:52:54:~/ros2_iron$ sudo apt update
Hit:1 http://packages.microsoft.com/repos/code stable InRelease
Hit:2 https://dl.google.com/linux/chrome/deb stable InRelease                  
Hit:3 http://us.archive.ubuntu.com/ubuntu jammy InRelease                      
Hit:4 http://packages.ros.org/ros2-testing/ubuntu jammy InRelease              
Get:5 http://us.archive.ubuntu.com/ubuntu jammy-updates InRelease [119 kB]     
Get:6 http://security.ubuntu.com/ubuntu jammy-security InRelease [110 kB]      
Get:7 http://us.archive.ubuntu.com/ubuntu jammy-backports InRelease [108 kB]   
Get:8 https://esm.ubuntu.com/apps/ubuntu jammy-apps-security InRelease [7,553 B]
Get:9 https://esm.ubuntu.com/apps/ubuntu jammy-apps-updates InRelease [7,459 B]
Hit:10 https://esm.ubuntu.com/infra/ubuntu jammy-infra-security InRelease
Hit:11 https://esm.ubuntu.com/infra/ubuntu jammy-infra-updates InRelease
Fetched 352 kB in 2s (167 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
habecker@2023-05-03 16:53:19:~/ros2_iron$ sudo apt install -y python3-rosdep
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-rosdep is already the newest version (0.22.2-1).
python3-rosdep set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
habecker@2023-05-03 16:53:28:~/ros2_iron$ sudo rosdep init
ERROR: default sources list file already exists:
	/etc/ros/rosdep/sources.list.d/20-default.list
Please delete if you wish to re-initialize
habecker@2023-05-03 16:53:43:~/ros2_iron$ cat /etc/ros/rosdep/sources.list.d/20-default.list
# os-specific listings first
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml osx

# generic
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
gbpdistro https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml fuerte

# newer distributions (Groovy, Hydro, ...) must not be listed anymore, they are being fetched from the rosdistro index.yaml instead
habecker@2023-05-03 16:53:59:~/ros2_iron$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
[... 138 lines cut ...]
[INFO] [1683158307.043998929] [talker]: Publishing: 'Hello World: 12'
[INFO] [1683158308.043967505] [talker]: Publishing: 'Hello World: 13'
[INFO] [1683158309.043973121] [talker]: Publishing: 'Hello World: 14'
[INFO] [1683158310.043988725] [talker]: Publishing: 'Hello World: 15'
[INFO] [1683158311.043973300] [talker]: Publishing: 'Hello World: 16'
[INFO] [1683158312.043903272] [talker]: Publishing: 'Hello World: 17'
^C[INFO] [1683158312.120960529] [rclcpp]: signal_handler(signum=2)
habecker@2023-05-03 16:58:32:~$ 
```

Listener:
```
habecker@2023-05-03 16:56:22:~$ . ~/ros2_iron/ros2-linux/setup.bash
ros2 run demo_nodes_py listener
[rti_connext_dds_cmake_module][warning] No RTI Connext DDS installation specified.. RTI Connext DDS will not be available at runtime,unless you already configured LD_LIBRARY_PATH manually.
[INFO] [1683158304.077305305] [listener]: I heard: [Hello World: 9]
[INFO] [1683158305.045858598] [listener]: I heard: [Hello World: 10]
[INFO] [1683158306.045828015] [listener]: I heard: [Hello World: 11]
[INFO] [1683158307.045855318] [listener]: I heard: [Hello World: 12]
[INFO] [1683158308.045810170] [listener]: I heard: [Hello World: 13]
[INFO] [1683158309.045757589] [listener]: I heard: [Hello World: 14]
[INFO] [1683158310.045819714] [listener]: I heard: [Hello World: 15]
^C^C
habecker@2023-05-03 16:58:31:~$ 
```

closed 2023-05-04 by Yadunund

cdhabecker, 2023-05-06:
**Problem: The given rosdep command would fail if the unpacked tar actually needed a ros dependency**

Reason: As per the instructions, the user does not `source ~/ros2_iron/ros2-linux/setup.bash` before running `rosdep`, which means that `rosdep` cannot query its database to convert raw package names such as `rclcpp` into distro-specific Debian package names such as `ros-iron-rclcpp`.

I'm working on a pull request.

This is the repro. In these examples, I change the documented `rosdep` command by removing the `--ignore-src` argument, which means that `rosdep` should find some unsatisfied dependencies.

**Clean shell**
```
habecker@2023-05-06 12:57:53:~$ printenv | grep -i ros
habecker@2023-05-06 12:58:07:~$ which rosdep
/usr/bin/rosdep
habecker@2023-05-06 12:59:12:~$ rosdep check --from-paths ros2_iron/ros2-linux/share --skip-keys "cyclonedds fastcdr fastrtps iceoryx_binding_c rmw_connextdds rti-connext-dds-6.0.1 urdfdom_headers"
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
All system dependencies have been satisfied
ERROR[ament_cmake_clang_format]: Cannot locate rosdep definition for [ament_cmake_test]
	rosdep key : ament_cmake_test
	OS name    : ubuntu
	OS version : jammy
	Data:
<no data>
ERROR[ament_cmake_copyright]: Cannot locate rosdep definition for [ament_copyright]
	rosdep key : ament_copyright
	OS name    : ubuntu
	OS version : jammy
	Data:
<no data>

```
**Single change: set ROS_DISTRO**
```
habecker@2023-05-06 13:03:11:~$ export ROS_DISTRO=iron
habecker@2023-05-06 13:03:19:~$ rosdep check --from-paths ros2_iron/ros2-linux/share --skip-keys "cyclonedds fastcdr fastrtps iceoryx_binding_c rmw_connextdds rti-connext-dds-6.0.1 urdfdom_headers"
System dependencies have not been satisfied:
apt	ros-iron-google-benchmark-vendor
apt	ros-iron-test-msgs
apt	ros-iron-ament-cmake-google-benchmark
...
habecker@2023-05-06 13:03:26:~$
```