osrf/ros2_test_cases #724: Installation instructions

Yadunund, 2023-04-05:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Chip: Arm64
- Os: Ubuntu Jammy

## Links
- [Installation instructions](https://docs.ros.org/en/rolling/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

tylerjw, 2023-05-03:
I'll take this one, following instructions here: https://docs.ros.org/en/iron/Installation/Alternatives/Ubuntu-Install-Binary.html

tylerjw, 2023-05-03:
I hit this issue when running the rosdep command in the tutorial:

```
$ rosdep install --from-paths ~/ros2_iron/ros2-linux/share --ignore-src -y --skip-keys "cyclonedds fastcdr fastrtps rti-connext-dds-6.0.1 urdfdom_headers"
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies (ROS distro is not set. Make sure `ROS_DISTRO` environment variable is set, or use `--rosdistro` option to specify the distro, e.g. `--rosdistro indigo`):
rmw_cyclonedds_cpp: Cannot locate rosdep definition for [iceoryx_binding_c]
```

tylerjw, 2023-05-03:
I tested and aside from the above issue this worked as expected.

clalancette, 2023-05-03:
> Should I add `iceoryx_binding_c` to the ignored keys?

I've already done it in https://github.com/ros2/ros2_documentation/pull/3493 , that one should be deployed soon.

clalancette, 2023-05-03:
That fix is now deployed, so closing this ticket out.

closed 2023-05-03 by clalancette