osrf/ros2_test_cases #725: Installation instructions

Yadunund, 2023-04-05:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Chip: Amd64
- Os: Ubuntu Jammy

## Links
- [Installation instructions](https://docs.ros.org/en/rolling/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

sea-bass, 2023-05-03:
Tackling this one.

sea-bass, 2023-05-03:
Worked great for me -- just a few small comments on the instructions.

First off, the note about `COLCON_IGNORE` could use some updates.
* Should we first suggest using `--packages-skip <name_of_package>`, as it doesn't modify the filesystem... and if people want to omit an entire folder containing multiple packages, we point them to `COLCON_IGNORE` as an option?
* I don't think it's worthwhile to draw the parallels to `CATKIN_IGNORE` at this point, but rather to just mention it directly.

Secondly, this page still has a section about the ROS 1 bridge documentation even though it's not officially supported. I guess the methodology presented in https://docs.ros.org/en/iron/How-To-Guides/Using-ros1_bridge-Jammy-upstream.html still works, so it might be better to link to this rather than the actual `ros1_bridge` repo?

clalancette, 2023-05-03:
> * Should we first suggest using `--packages-skip <name_of_package>`, as it doesn't modify the filesystem... and if people want to omit an entire folder containing multiple packages, we point them to `COLCON_IGNORE` as an option?

Yeah, I agree.  We should just have people use `--packages-skip`.  Here's a PR that makes the change: https://github.com/ros2/ros2_documentation/pull/3500

> Secondly, this page still has a section about the ROS 1 bridge documentation even though it's not officially supported. I guess the methodology presented in [...]

That's a good idea.  Here's a PR that does this: https://github.com/ros2/ros2_documentation/pull/3501

Just to double-check: did you use the installation page from http://docs.ros.org/en/iron/Installation/Ubuntu-Install-Debians.html for this testing?  Your comments about COLCON_IGNORE makes me think you used the from-source instructions, but I might be missing something.

sea-bass, 2023-05-03:
Thanks for the doc PRs!

I did a source install at first, but there was no issue for source builds on jammy. I did also do a debian install, but agree my comments on COLCON_IGNORE certainly don't apply there.

fujitatomoya, 2023-05-04:
/assign @fujitatomoya

fujitatomoya, 2023-05-04:
### main installation procedure

- https://docs.ros.org/en/iron/Installation/Ubuntu-Install-Debians.html

### result

confirmed that https://docs.ros.org/en/iron/Installation/Ubuntu-Install-Debians.html#try-some-examples works w/o problems

Yadunund, 2023-05-04:
I'll close this since we have multiple confirmations

closed 2023-05-04 by Yadunund