osrf/ros2_test_cases #726: Installation instructions

Yadunund, 2023-04-05:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Chip: Arm64
- Os: Ubuntu Jammy

## Links
- [Installation instructions](https://docs.ros.org/en/rolling/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

fujitatomoya, 2023-05-04:
/assign @fujitatomoya

fujitatomoya, 2023-05-04:
### main installation procedure

- https://docs.ros.org/en/iron/Installation/Ubuntu-Install-Debians.html

### result

confirmed that https://docs.ros.org/en/iron/Installation/Ubuntu-Install-Debians.html#try-some-examples works w/o problems

Yadunund, 2023-05-04:
thanks for testing!

closed 2023-05-04 by Yadunund