osrf/ros2_test_cases #727: Installation instructions

Yadunund, 2023-04-05:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- DDS vendor: FastDDS
- BuildType: Source
- Chip: Amd64
- Os: Rhel 9

## Links
- [Installation instructions](https://docs.ros.org/en/rolling/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

odra, 2023-05-04:
I would like to take this issue if nobody has yet :)

Yadunund, 2023-05-05:
Great, let us know how it goes!

pradyum, 2023-05-06:
**Setup**

- DDS vendor: FastDDS
- BuildType: Source
- Chip: Amd64
- OS: Rhel 9

1. I encountered a problem while running 

``sudo dnf config-manager --set-enabled crb``

*Output*

``Error: No matching repo to modify: crb``

Alternatively to enable crb i used the following command

``sudo crb enable``

2. I have to build it three time to achieve successful build. The outputs of all three build are attached

[build1 fail.txt](https://github.com/osrf/ros2_test_cases/files/11412314/build1.fail.txt)
[build2 stderr output.txt](https://github.com/osrf/ros2_test_cases/files/11412315/build2.stderr.output.txt)
[build3 success.txt](https://github.com/osrf/ros2_test_cases/files/11412316/build3.success.txt)

Other-than these issue the build using the instruction is successful. And the demo is attached below

[image]

odra, 2023-05-07:
RHEL9 should the steps in this section: https://docs.fedoraproject.org/en-US/epel/#_rhel_9

odra, 2023-05-07:
Platform (RHEL9.1 VM):

```
Red Hat Enterprise Linux 9.1 (Plow)
Kernel 5.14.0-162.6.1.el9_1.x86_64 on an x86_64
```

The build failed at first but it worked when skiping examples:

```
colcon build --symlink-install --packages-skip image_tools intra_process_demo
```

I was able to run the talker-listener example:
[image]

pradyum, 2023-05-07:
Yeah I followed the step in https://docs.fedoraproject.org/en-US/epel/#_rhel_9 to add epel repository and used ```sudo crb enable``` to enable crb repository.

I found the reason why my first build failed. It is due to ```c++: fatal error: Killed signal terminated program cc1plus``` which is caused by not having enough memory in my computer. In my first build the computer ran out of memory. While doing second time 346 packages were already built and rest used less system memory to build. 

There is no problem with the packages failing to build. The ```--packages-skip``` is used to skip one or more packages. In the example it is shown on skipping opencv library.

Yadunund, 2023-05-08:
Thanks for testing! I'll consider this done

closed 2023-05-08 by Yadunund