osrf/ros2_test_cases #728: Installation instructions

Yadunund, 2023-04-05:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- DDS vendor: FastDDS
- BuildType: Source
- Chip: Amd64
- Os: Windows

## Links
- [Installation instructions](https://docs.ros.org/en/rolling/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

sea-bass, 2023-05-03:
Looking at this issue today. 

FYI it seems that there is no assignee access for random users right now -- might be beneficial to change these settings so assignees don't have to be moderated by OR.

BorisBoutillier, 2023-05-04:
I have tried to install on windows from source.
Following the instruction I have skipped the dependencies installation until the 'Install additional dependencies from Chocolatey' as I had already installed them to test the binary installation.

`colcon build` step failed on `mimick_vendor` compilation,  as is already reported in https://github.com/ros2/ros2/issues/1368 
So while the binary windows installation works with Windows 11 ( even if only Windows 10 is officially supported), building from source is effectively not supported for Windows 11.

We will need another person with Windows 10 to test build from source.

Yadunund, 2023-05-04:
Source build in-progress on my windows 10 machine. 1hr 20mins in with 12 more packages to go 🙃

Yadunund, 2023-05-04:
Alright no issues with my build + testing so I'll go ahead and close this one 
[image]

closed 2023-05-04 by Yadunund