osrf/ros2_test_cases #736: Launch testing

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Launch testing examples](https://github.com/ros2/launch/blob/master/launch_testing/README.md#examples)

## Checks
- [ ] **Test `hello_world_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/hello_world_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `good_proc_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/good_proc_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `terminating_proc_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/terminating_proc_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `args_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Show the arguments
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/args_launch_test.py --show-args
       ```
    1.
       Run the test
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/args_launch_test.py dut_arg:=hey
       ```
  
    
  </details>
- [ ] **Test `context_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/context_launch_test.py
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-08:
Hello, could you please assign this task to me? Thank you so much.

juliamp22, 2023-05-08:
**Test hello_world_launch_test.py**

I directly execute the` launch_test test/launch_testing/examples/hello_world_launch_test.py` 

but it says that the Test file does not exist, do I have to install some package? Thank you so much. 

[image]

Yadunund, 2023-05-09:
You're passing a local path to launch_test which does not exist. The testing instructions should be improved to make things clearer. Could you try

```
git clone https://github.com/ros2/launch.git
cd launch/launch_testing
call PATH-TO-BINARY-UNPACK\setup.bat # where you unpacked the binaries 
# then run the launch_test commands
```

juliamp22, 2023-05-09:
Hello @Yadunund you are totally right, now works perfectly.

**hello_world_launch_test.py**

[image]

juliamp22, 2023-05-09:
Test **good_proc_launch_test.py**, works fine:

[image]

juliamp22, 2023-05-09:
Test **terminating_proc_launch_test.py**, appears some errors in testing, however, seems that works fine:

[image]

juliamp22, 2023-05-09:
Test **args_launch_test.py**, works fine:

[image]

juliamp22, 2023-05-09:
Test **context_launch_test.py**, appears some erros:

[image]

juliamp22, 2023-05-09:
Hello @Yadunund there appears some errors during the test, how do you want to proceed? Thank you so much.

Yadunund, 2023-05-10:
Thanks for testing! This can be considered a success since I see OKs for all the tests. The errors are intentional as the tests are sending signals to terminate the process.

closed 2023-05-10 by Yadunund