osrf/ros2_test_cases #737: Launch testing

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Launch testing examples](https://github.com/ros2/launch/blob/master/launch_testing/README.md#examples)

## Checks
- [ ] **Test `hello_world_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/hello_world_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `good_proc_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/good_proc_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `terminating_proc_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/terminating_proc_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `args_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Show the arguments
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/args_launch_test.py --show-args
       ```
    1.
       Run the test
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/args_launch_test.py dut_arg:=hey
       ```
  
    
  </details>
- [ ] **Test `context_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/context_launch_test.py
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-09:
On a debian jammy installation with fastdds.

I have run the 5 `launch_test` commands.

All Ok.

But if possible I would suggest for future release testcase to update the command to properly point to the installed test path, as this is a debian install, so path is known. (`/opt/ros/iron/share/launch_testing`)
Namely
```
# User input in terminal 1
launch_test  /opt/ros/iron/share/launch_testing/examples/hello_world_launch_test.py
```
```
# User input in terminal 1
launch_test /opt/ros/iron/share/launch_testing/examples/good_proc_launch_test.py
```
```
# User input in terminal 1
launch_test /opt/ros/iron/share/launch_testing/examples/terminating_proc_launch_test.py
```
```
# User input in terminal 1
launch_test /opt/ros/iron/share/launch_testing/examples/args_launch_test.py --show-args
```
```
# User input in terminal 1
launch_test /opt/ros/iron/share/launch_testing/examples/args_launch_test.py dut_arg:=hey
```
```
# User input in terminal 1
launch_test /opt/ros/iron/share/launch_testing/examples/context_launch_test.py
```

[image]
[image]
[image]
[image]
[image]

Yadunund, 2023-05-11:
Thanks for testing this. Agreed that the test case needs to be updated and I've taken note.

closed 2023-05-11 by Yadunund