osrf/ros2_test_cases #738: Launch testing

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Launch testing examples](https://github.com/ros2/launch/blob/master/launch_testing/README.md#examples)

## Checks
- [ ] **Test `hello_world_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/hello_world_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `good_proc_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/good_proc_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `terminating_proc_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/terminating_proc_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `args_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Show the arguments
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/args_launch_test.py --show-args
       ```
    1.
       Run the test
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/args_launch_test.py dut_arg:=hey
       ```
  
    
  </details>
- [ ] **Test `context_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/context_launch_test.py
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a Windows11 binary installation.

I have run the test_cases, using the installed version of the examples tests, in `ros2_iron\share\launch_testing\examples`

All 5 tests runned as expected, without ERRORS

[image]

Yadunund, 2023-05-05:
Great, thank you!