osrf/ros2_test_cases #739: Launch testing

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Launch testing examples](https://github.com/ros2/launch/blob/master/launch_testing/README.md#examples)

## Checks
- [ ] **Test `hello_world_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/hello_world_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `good_proc_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/good_proc_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `terminating_proc_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/terminating_proc_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `args_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Show the arguments
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/args_launch_test.py --show-args
       ```
    1.
       Run the test
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/args_launch_test.py dut_arg:=hey
       ```
  
    
  </details>
- [ ] **Test `context_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test test/launch_testing/examples/context_launch_test.py
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Kartik9250, 2023-05-06:
Hey, I would like to work on this ticket

Kartik9250, 2023-05-06:
The commands did not work at first, would have been due to missing directory so I clone the `ros2/launch` repository and then used the commands give above inside that directory.

### Checks
- [x] `Test hello_world_launch_test.py`
[image]

- [x] `Test good_proc_launch_test.py`
[image]

- [x] Test `terminating_proc_launch_test.py`
    - This error was prompted while running this test case:
   ```
    [terminating_proc.py-2] Ready
    [terminating_proc.py-2] Called with arguments ['--exception']
    [terminating_proc.py-2] Traceback (most recent call last):
    [terminating_proc.py-2]   File "/opt/ros/iron/lib/launch_testing/terminating_proc.py", line 33, in <module>
    [terminating_proc.py-2]     raise Exception('Process had a pretend error')
    [terminating_proc.py-2] Exception: Process had a pretend error 
    [ERROR] [terminating_proc.py-2]: process has died [pid 8851, exit code 1, cmd '/usr/bin/python3 
     /opt/ros/iron/lib/launch_testing/terminating_proc.py --exception'].
     ok
     ```

    ![image](https://user-images.githubusercontent.com/58456392/236613285-903b5f24-2e1d-4ada-b35e-0de3de3d539f.png)

- [x] Test `args_launch_test.py`
    - Show the arguments
        ![image](https://user-images.githubusercontent.com/58456392/236613527-6d9418de-9390-436f-8a5f-a463e5eb10ee.png)
    - Run the test
         ![image](https://user-images.githubusercontent.com/58456392/236613612-0ab65a90-64b3-4a3a-b800-052877681215.png)

- [x] Test `context_launch_test.py`
![image](https://user-images.githubusercontent.com/58456392/236613787-4822e08c-140b-4b4c-8086-0a797ede030c.png)

Yadunund, 2023-05-08:
Thanks for testing!

closed 2023-05-08 by Yadunund