osrf/ros2_test_cases #742: Lifecycle demo

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/demos/blob/master/lifecycle/README.rst)

## Checks
- [ ] **Run files individually**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       [image]
       ```bash
       # User input in terminal 1
       ros2 run lifecycle lifecycle_talker
       ```
    1.
       [image]
       ```bash
       # User input in terminal 2
       ros2 run lifecycle lifecycle_listener
       ```
    1.
       [image]
       ```bash
       # User input in terminal 3
       ros2 run lifecycle lifecycle_service_client
       ```
  
    **Expect**
    
    1.
       The images match.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-08:
# Setup

- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

[image]

# Result

[image]

[image]

**I am getting the following outputs. The Lifecycle demo is working.**

Yadunund, 2023-05-08:
Thanks for testing!

closed 2023-05-08 by Yadunund