osrf/ros2_test_cases #749: Local workspace overlay

Yadunund, 2023-04-05:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- DDS vendor: Connext
- BuildType: Debian
- Chip: Amd64
- Os: Ubuntu Jammy

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

tylerjw, 2023-05-03:
When trying to build an underlay workspace with context_dds installed I get this error:

```
[ERROR] [1683132753.341107851] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_connextdds'), with error message 'failed to load shared library 'librmw_connextdds.so' due to dlopen error: libnddscd.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99, at ./src/functions.cpp:65', exiting with 1., at ./src/rcl/rmw_implementation_identifier_check.c:139
```

I installed context dds: `sudo apt install ros-iron-rmw-connextdds`
Set the rmw implementation like this: `export RMW_IMPLEMENTATION=rmw_connextdds`

tylerjw, 2023-05-03:
Underlay:
```
$ vcs export
repositories:
  RSL:
    type: git
    url: git@github.com:PickNikRobotics/RSL.git
    version: main
  cpp_polyfills:
    type: git
    url: git@github.com:PickNikRobotics/cpp_polyfills.git
    version: main
```

Overlay (`generate_parameter_library` depends on `RSL`):
```
repositories:
  generate_parameter_library:
    type: git
    url: git@github.com:PickNikRobotics/generate_parameter_library.git
    version: main
```

Steps tested:
1. Build underlay based on iron (`colcon build`)
2. Run tests in underlay (`colcon test`)
3. Source underlay workspace (`source install/setup.bash`)
4. Build overlay workspace
5. Test overlay workspace

OS: Ubuntu Jammy 22.04

clalancette, 2023-05-03:
Thanks for testing!

closed 2023-05-03 by clalancette